VoronDesign/Voron-Hardware #13: Question about Hall_Effect_XY_Endstop development status

Lefuneste83, 2020-05-07:
Hello. Before sending the PCB to fab and while waiting for the V2.4 manual to be published, I was wondering what is the current status of development of the Hall_Effect_XY_Endstop project. Is it sufficiently stabilized to be manufactured ? Or do you have some clues about future evolution worth waiting for ? Thanks for the info.

timmit99, 2020-05-08:
@Lefuneste83 there are no current projects working on the Hall Effect. It's a great board in a stable design with no changes planned for the future. I was planning to add an interactive BOM similar to [this one](http://htmlpreview.github.io/?https://github.com/VoronDesign/Voron-Hardware/blob/master/Klipper_Expander/KiCad/KlipperExpander_iBOM.html) as well as doing some checks on current part status but that wont change the PCB.

timmit99, 2020-05-08:
@Lefuneste83  I was just informed that there was some talk about adding a silkscreen. to the output connector. I can add that in and resubmit to github today.

Lefuneste83, 2020-05-08:
Hello Tim and thank you for the update. I'll get the newer Kicad design and send it to Oshpark then. Don't forget to regenerate the associated gerber files if people want to directly send it to fab. For the BOM all parts are already ready to order at my suppliers.  The interactive BOM is very nice for the expander.

timmit99, 2020-05-08:
@Lefuneste83  the Gerbers and silkscreen changes have been done. sorry about the delay, had some issues with my git repo for some reason.

john--, 2020-05-08:
Linked to #18

closed 2020-05-08 by john--

Lefuneste83, 2020-05-09:
Hello Tim. Please no need to apologize my friend. It's really very kind of you to reply so fast. I have put the design on Oshpark for those interested in having it manufactured.

https://oshpark.com/shared_projects/YbqMi5lD

I highly recommend Oshpark service even for those living abroad like I do. Their quality is equal to none and their service super reliable. It is a bit more expensive than Chinese PCB fabs, but for such small quantity it does not matter. I'll manufacture a dozen of these boards and let the community know in case someone needs one. All the best !