VoronDesign/Voron-Hardware #72: Remove JLC text from Microswitch_endstop B.silks

bigjoe4265, 2021-03-07:
(gr_text JLCJLCJLCJLC (at 167.6 68.4) (layer B.SilkS)
    (effects (font (size 1 1) (thickness 0.15)) (justify mirror))
  )

raymondh2, 2021-03-07:
@randellhodges

timmit99, 2021-03-07:
@bigjoe4265 that line is there so when the board is ordered from JLC, they out the order number there so it's in a known location. 
If the original designer wants to remove it that's fine, but its a nice addition for making boards.

bigjoe4265, 2021-03-07:
> 
> 
> @bigjoe4265 that line is there so when the board is ordered from JLC, they out the order number there so it's in a known location.
> If the original designer wants to remove it that's fine, but its a nice addition for making boards.

that's great, but what if I wish to to use another pcb manufactorer?

raymondh2, 2021-03-07:
Like timmit suggested you can remove it yourself.

bigjoe4265, 2021-03-07:
> 
> 
> Like timmit suggested you can remove it yourself.

I'm not that familiar with KiCad.  I tried that approach and can't get it to update the files properly.

timmit99, 2021-03-07:
@bigjoe4265 let me make a version without the jlc text. Do you want me to screen capture to show you how? Any interest in learning kicad basics?

bigjoe4265, 2021-03-07:
Yes, but I don't want to clutter up the reported issues here.  I was able to update the schematic file however it's saying I'm missing libraries when I try to update pcb.

timmit99, 2021-03-07:
Let me take a look and report back!

timmit99, 2021-03-07:
@bigjoe4265 here's what i was able to do.
https://youtu.be/DSm7_0XXC-4

[Gerbers_No_JLC.zip](https://github.com/VoronDesign/Voron-Hardware/files/6096488/Gerbers_No_JLC.zip)

bigjoe4265, 2021-03-07:
Thank you for the assistance and the tutorial.

closed 2021-03-07 by bigjoe4265