VoronDesign/Voron-Hardware #87: Pins swapped on HALL-EFFECT Endstop

th0mpy, 2021-06-23:
Looking at your schematic, you have pins 5 and 6 swapped on the LM. It'll work but the swap will cause it to go low all of the time, unless it detects a magnet. If you like I can send over the updated schematics.

timmit99, 2021-06-23:
The nature of the device is to pull the signal low nominally and only go up to 3.3V when triggered.
@thebrakshow that is correct right? This schematic hasn't changed since the OG version.

th0mpy, 2021-06-23:
The X triggers in the opposite direction from the Y. The X axis has the sensor going into pin 2, which is the negative input. The Y has it going into pin 5, which is the positive input on the second comparator. That means the endstop light is on at all times, except when triggered. I did not think this was desired, but either way, they are opposite from each other

troy-jacobson, 2021-10-17:
I am seeing the same behavior and concur with th0mpy.   The schematic shows the sensors going to inputs IN_A1 and IN_B1, but the corresponding pins on the device are Input A- and Input B+.  Probing the board with the multimeter confirms the connections between the sensors and comparators.

timmit99, 2021-12-30:
I'm looking to update this with a minor revision to the hall effect to also remove the old connector and replace with a right angle JST-XH which is much more common on printers now.