VoronDesign/Voron-Hardware #112: Update M300 macro for STM32_Mini12864

solrac8874, 2021-11-21:
I get a parameter_S value not value error when restarted klipper with the includes klipper-mini12864.cfg

I think Klipper gcode parameter syntax changed recently.

Same error I previously with the same syntax for the M300 macro when I had it in my printer.cfg before installing the STM32_Mini12864 adapter.

In klipper-mini12864.cfg, instead of:

[gcode_macro M300]
default_parameter_S=1000
default_parameter_P=100
gcode:  SET_PIN PIN=BEEPER_pin VALUE={S}
        G4 P{P}
        SET_PIN PIN=BEEPER_pin VALUE=0

The M300 gcode should read something like:

[gcode_macro M300]
gcode:
    # Use a default 1kHz tone if S is omitted.
    {% set S = params.S|default(1000)|int %}
    # Use a 10ms duration is P is omitted.
    {% set P = params.P|default(100)|int %}
    SET_PIN PIN=beeper_pin VALUE={S} CYCLE_TIME={ 1.0/S if S > 0 else 1 }
    G4 P{P}
    SET_PIN PIN=beeper_pin VALUE=0

I can't claim credit for the gcode M300 code, I found it on the Voron discord when I was getting the error when I had my Mini112864 directly connected to the Z_MCU (Voron2).  I seems to work with the revised gcode; at least I don't get a Klipper error for an invalid parameter S value.

john--, 2021-11-21:
Feel free to submit a PR with the change!

solrac8874, 2021-11-21:
I submitted PR#113.  Hope I did the PR correctly.

solrac8874, 2021-11-30:
PR#113 commit merged

closed 2021-11-30 by solrac8874