VoronDesign/Voron-Hardware #128: Klipper Expander restart not working

RcTomcat1, 2022-01-23:
Hi,

I just installed my klipper expander. Flashed with current klipper.
Issue:
Issuing a restart command via fluidd leads to the expander board being unreachable. I need to power cycle the board or press the physical reset button.

Expected behaviour: I expected the restart command to just restart the MCUs (board and expander)

Setup:
FYSETC Spider 2.2 on a Voron 2.4
Raspberry Pi 3b+ running Klipper v0.10.0-235-gf97fd7c6  and fluidd v.1.16.2
Klipper Expander with flashed with Klipper via the Pi.

Config:
`
[mcu expander]
serial:  /dev/serial/by-id/usb-Klipper_stm32f042x6_2A002E000843534133343520-if00
restart_method: command
#baud: 250000
#[output_pin mosfet0]
#pin: expander:PA0
#value: 0
#shutdown_value: 0

[temperature_sensor raspberry_pi]
sensor_type: temperature_host
min_temp: 10
max_temp: 100

[temperature_sensor MCU_Board_Expander]
sensor_type: temperature_mcu
min_temp: 10
max_temp: 100

[heater_fan controller_fan-expander]
##	Controller fan - FAN2 Connector
pin: expander:PA0
kick_start_time: 0.5
heater: heater_bed
heater_temp: 45.0
max_power: 0.5
shutdown_speed: 0.0
kick_start_time: 5.0
fan_speed: 1.0

#[output_pin mosfet1]
#pin: expander:PA1
#value: 0
#shutdown_value: 0

[heater_fan exhaust_fan]
##  Exhaust fan
pin: expander:PA1
kick_start_time: 0.5
heater: heater_bed
heater_temp: 60.0
max_power: 0.5
shutdown_speed: 0.0
kick_start_time: 5.0
fan_speed: 1.0
#max_power: 0.5
#off_below: 0.10
#[output_pin mosfet2]
#pin: expander:PA2
#value: 0
#shutdown_value: 0

#[output_pin mosfet3]
#pin: expander:PA3
#value: 0
#shutdown_value: 0

# Status LED lights when klipper connects
[static_digital_output onboardLED]
pins: !expander:PA4

[neopixel Pixel]
pin: expander:PB1
chain_count: 1
initial_RED: 0.9
initial_GREEN: 0.3
initial_BLUE: 0.0

#[display]
#lcd_type: sh1106
#i2c_bus: i2c1a

# Other Pins
# GPIO = PA7

#[temperature_sensor t0]
#sensor_type: ATC Semitec 104GT-2
#sensor_pin: expander:PA5
#gcode_id=0

[temperature_sensor Gehaeuse]
sensor_type: ATC Semitec 104GT-2
sensor_pin: expander:PA6
gcode_id=1`

This config is referenced in the printer.cfg

Is it normal not to be able to restart via command?

raymondh2, 2022-02-14:
This seems to be an issue with klipper and the MCU used. I suggest seeking help on the Klipper help sites

closed 2022-02-14 by raymondh2

HitLuca, 2023-11-04:
@RcTomcat1 have you ever fixed your issue?