VoronDesign/Voron-Hardware #134: Shutdown V0 display

Schnabulation, 2022-02-15:
Dear VoronDesign

I have a quite specific setup whereas OctoPrint and Klipper run on a virtual machine that is always powered. The MCU, webcam and the V0 display are all connected to the server and are recieving power all the time. I have a plugin that shuts down a smart switch so the printer itself loses power.

However: the V0 display is still powered on. Is it possible to issue a command so that the display powers down and goes dark?

VinnyCordeiro, 2022-02-15:
> The MCU, webcam and the V0 display are all connected to the server and are recieving power all the time.

You answered yourself here. The V0 Display is a very basic USB device and does not have a shutdown state defined, so the only way I see to turn it off is by actively removing power from it. That could be done in many ways. Does your server have a parallel port, the type used for old printers? If yes, you could use it to turn on and off the display through a MOSFET or even a relay. If not, then you'll have to look after more sophisticated solutions.

Schnabulation, 2022-02-15:
Thank you for your reply.

What I noticed however is that the V0 display does not boot until Klipper tells it to - even if it has USB-power. Meaning that when I unplug the display and plug it back in, it stays black until I reboot Klipper - then it comes back to life. So I was hoping that maybe it waits for a command to start and could therefore also be told to shut down.

raymondh2, 2022-02-16:
My guess as this is a behaviour that can't be changed. You are more likely to find an awnser at the Klipper help sites. https://www.klipper3d.org/Contact.html