VoronDesign/Voron-Hardware #154: STM32_Mini12864 Shutdown Kill_Switch Problem

wissencat1, 2022-05-04:
Hello i have the Problem that i get random Kill switch Problems with my Display and the STM32_Mini12864 board.

I did not Self source it. I bought it from Fysetc.

I did invert then the Kill Switch Button.

Aswell when i use the beeper it detects the kill_switch input and shuts down.

I have no R4 Resistor on my display for the kill switch, but an R3

[image]

Is there an Shematic  as PDF of the Pins of the Labeling and Naming for the board?

[display]
lcd_type: uc1701

kill_pin: !menu:PF0     my setting:    kill_pin: menu:PF0

spi_bus: spi1
cs_pin: menu:PA4
a0_pin: menu:PA2
rst_pin: menu:PA3
contrast: 63
encoder_pins: ^menu:PA0, ^menu:PA1
click_pin: ^!menu:PB1

Thanks in Advance

closed 2022-06-06 by wissencat1

Dymental, 2022-07-07:
Did you solve your Problem?
I have exact the same configuration and I can not get the hardware reset button on the display to work, would like to configure it as Emergency stop
My Hardware:
BTT Mini12864
Fystec SMT32 Adapter
Raspberry 4b
BTT SKR Pico

wissencat1, 2022-07-08:
Sadly not button is without Funktion so I used the reset button of the board as an emergency stop that kills the firmware and stops the printer because the mcu is offline and klipper stops then.