VoronDesign/Voron-Hardware #159: typo in the commands for firmware flashing?

PMvE, 2022-05-28:
it looks like there is a typo in the klipper expander instalation guide under Setup it says to use 
"CD ~/klipper"
to go to the klipper directory it returns 
"-bash: CD: command not found" 
in the voron documentation it says 
"cd ~/klipper" 
that one works. i am really bad with this stuff so i might be wrong.

raymondh2, 2022-05-28:
@timmit99

timmit99, 2022-05-28:
Yep, I sword I had fixed that.... Something must have reverted it.
@PMvE do you know how to make a pull request? Imm away from my PC for a bit so I can't fix it right away.

PMvE, 2022-05-28:
i think google can help me with that and i should learn it anyway

PMvE, 2022-05-28:
https://github.com/VoronDesign/Voron-Hardware/pull/160 I think i did it wrong sorry

timmit99, 2022-05-28:
@PMvE looks good to me.

@raymondh2 thoughts?

commit ed1b40ea (2022-06-06): Update README.md (#160)

fixed a typo (cd ~/klipper
Closes #159
closed 2022-06-06 by raymondh2