VoronDesign/Voron-Hardware #179: V0-Umbilical: No cutout for neopixel wire routing

vavaroutsos, 2022-09-29:
I don't know how people have been routing the neopixel wiring, but the only way I see to get from the rear electronics area to the top of the build chamber is through the Y end stop switch cutout. This is too small for anything more than one set of neopixel wires. I think this cutout should be enlarged or additional cutouts should be added for neopixel wiring. Am I missing something?

dajones89, 2022-11-26:
@vavaroutsos Not sure if this is too late but If you are carefull you can cut a notch out of the pcb, I used a hacksaw, I mirrored the notch for the endstop on the other side, Have on the left LED wiring, on the right LED wiring and Endstop wiring.

vavaroutsos, 2022-11-26:
I modified the cover and added two holes for the neopixel connectors. Then I de-soldered them and installed them on the topside. I think this is the right solution if they are to be for case lights. Now I can easily unplug them if I need to remove the hinged top hat for any reason. Attached is the modified cover if you want to include it in the repo.

 

Best Regards,

 

Pete