VoronDesign/Voron-Hardware #193: 0.2 version of PCB mount sets PCB too low

EricZimmerman, 2023-01-27:
This part:

https://github.com/VoronDesign/Voron-Hardware/blob/master/V0-Umbilical/STLs/V0.2/V0.2_Toolhead_PCB_Mount.STL

has minimal clearance with the BA motor mounts in the back. seems like the pcb should be moved up as far as allows for hole spacing to provide more clearance with mounts

EricZimmerman, 2023-01-27:
https://discord.com/channels/460117602945990666/710952853514223617/1068596392966230178

kejar31, 2023-02-04:
I believe this was resolved with my update, correct ?

Gabbajoe, 2023-03-22:
Hi no for me it is stil la issue that the [V0.2_Toolhead_PCB_Mount.STL](https://github.com/VoronDesign/Voron-Hardware/blob/master/V0-Umbilical/STLs/V0.2/V0.2_Toolhead_PCB_Mount.STL) is not clearing the A/B Motor upper frames see:
[image]

I did now the lazy way an shave off in an angle plastic unter the screw holes.

i8086m, 2023-03-30:
Having same issue with [V0.2_Toolhead_PCB_Mount.STL](https://github.com/VoronDesign/Voron-Hardware/blob/master/V0-Umbilical/STLs/V0.2/V0.2_Toolhead_PCB_Mount.STL) colliding

[image]

tkalbitz, 2023-04-24:
Had the same issue and shaved also the screw feature off.

closed 2023-07-18 by timmit99