VoronDesign/Voron-Hardware #201: STL/STEP files for Daylight/Disco Stick 1515 brackets do not work

pdsavas, 2023-03-30:
I just downloaded the STEP and STL files for the 1515 brackets and tried to open them with several different applications, but the files appear to be corrupt. Would someone be able to correct this and reupload working copies, please?

Thanks,
pete

BerzerkaDurk, 2023-04-05:
Guessing you just pressed the download button in the STL folder. Instead, right-click the button and choose "save link as" or "save target as", depending on your browser.

pdsavas, 2023-04-06:
No, I actually did right-click and try to save the files using the "save target as," and they always came down corrupted. I couldn't get any slicer or 3D modeling program to open them.

timmit99, 2023-04-06:
Can you upload one of those files here for us to check?

Thanks!

pdsavas, 2023-04-06:
[Brackets_1515_x2.zip](https://github.com/VoronDesign/Voron-Hardware/files/11166162/Brackets_1515_x2.zip)
There you go.

pete

timmit99, 2023-04-06:
> [Brackets_1515_x2.zip](https://github.com/VoronDesign/Voron-Hardware/files/11166162/Brackets_1515_x2.zip) There you go.
> 
> pete

You downloaded the webpage, not the STL file.
[image]

timmit99, 2023-04-06:
"Save link as" is what you want here.
[image]

That gives me this file which works:
[image]

pdsavas, 2023-04-06:
I swear that's what I did... (sigh). Thanks, and sorry.

BerzerkaDurk, 2023-04-06:
Called it. :o)
99/100 times when someone downloads a corrupt STL from the repo, that's it.

pepegragera, 2023-07-22:
The files are corrupted, does anyone have them? the links that you have put are downloaded but when uploading to the laminator it says that it is corrupt

closed 2023-08-09 by nemgrea