VoronDesign/Voron-Hardware #222: Firmware update fails to flash, will not "fit"

kingozone, 2024-03-16:
Currently running Klipper on Ender 3 V3 SE. Recently updated, and needed to update the Exapnder board. When I try to update it, I get the following error:

biqu@BTT-CB1:~/klipper$ make clean
biqu@BTT-CB1:~/klipper$ make flash FLASH_DEVICE=0483:df11
  Creating symbolic link out/board
  Building out/autoconf.h
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/initial_pins.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/trsync.o
  Compiling out/src/dirzctl.o
  Compiling out/src/hx711s.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/neopixel.o
  Compiling out/src/pulse_counter.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/spi_software.o
  Compiling out/src/thermocouple.o
  Compiling out/src/sensor_adxl345.o
  Compiling out/src/sensor_angle.o
  Compiling out/src/sensor_mpu9250.o
  Compiling out/src/sensor_bulk.o
  Compiling out/src/stm32/watchdog.o
  Compiling out/src/stm32/gpio.o
  Compiling out/src/stm32/clockline.o
  Compiling out/src/stm32/dfu_reboot.o
  Compiling out/src/generic/crc16_ccitt.o
  Compiling out/src/generic/armcm_boot.o
  Compiling out/src/generic/armcm_irq.o
  Compiling out/src/generic/armcm_reset.o
  Compiling out/src/../lib/stm32f0/system_stm32f0xx.o
  Compiling out/src/generic/timer_irq.o
  Compiling out/src/stm32/stm32f0_timer.o
  Compiling out/src/stm32/stm32f0.o
  Compiling out/src/stm32/gpioperiph.o
  Compiling out/src/stm32/stm32f0_adc.o
  Compiling out/src/stm32/stm32f0_i2c.o
  Compiling out/src/stm32/spi.o
  Compiling out/src/stm32/usbfs.o
  Compiling out/src/stm32/chipid.o
  Compiling out/src/generic/usb_cdc.o
  Building out/compile_time_request.o
Version: 25778772
  Preprocessing out/src/generic/armcm_link.ld
  Linking out/klipper.elf
/usr/lib/gcc/arm-none-eabi/8.3.1/../../../arm-none-eabi/bin/ld: out/klipper.elf section `.text' will not fit in region `rom'
/usr/lib/gcc/arm-none-eabi/8.3.1/../../../arm-none-eabi/bin/ld: region `rom' overflowed by 10512 bytes
collect2: error: ld returned 1 exit status
make: *** [Makefile:72: out/klipper.elf] Error 1

How can I fix this? I have followed the manual and everytime, I get this error.

timmit99, 2024-03-16:
Can you share screenshots of your menu config and advanced options please?

kingozone, 2024-03-19:
[image]
[image]

kingozone, 2024-03-23:
After buying a second one, I'm getting the same results

kingozone, 2024-03-27:
Any ideas?

yardi83, 2025-02-09:
I was running into the exact same issue today while trying to update the expander from klipper v0.11XX to v0.12

This settings fixed it for me:
https://github.com/VoronDesign/Voron-Hardware/issues/231

Docs should be updated with this