VoronDesign/Voron-Hardware #223: Termistor T0 configuration

pepegragera, 2024-05-26:
Hello, I am trying to control a thermistor for the inside of the machine, I have installed the thermistor in T0 but klipper tells me the following after adding the line to my klipperExpander.cfg file

[temperature_sensor t0]
sensor_type: ATC Semitec 104GT-2
sensor_pin: PA6
gcode_id: T0
[klippy.log](https://github.com/VoronDesign/Voron-Hardware/files/15448510/klippy.log)

**klipperExpander.cfg is:**
[image]

[mcu expander]
serial: /dev/serial/by-id/usb-Klipper_stm32f042x6_100002000543304846333520-if00
restart_method: command

[temperature_sensor t0]
sensor_type: ATC Semitec 104GT-2
sensor_pin: PA6
gcode_id: T0

#[output_pin mosfet0]
#pin: expander:PA0
#pwm: true
#cycle_time: 0.010
#value: 0
#shutdown_value: 0

#[output_pin mosfet1]
#pin: expander:PA1
#pwm: true
#cycle_time: 0.010
#value: 0
#shutdown_value: 0

#[output_pin mosfet2]
#pin: expander:PA2
#pwm: true
#cycle_time: 0.010
#value: 0
#shutdown_value: 0

#[output_pin mosfet3]
#pin: expander:PA3
#pwm: true
#cycle_time: 0.010
#value: 0
#shutdown_value: 0

# Status LED lights when klipper connects 
#[static_digital_output onboardLED]
#pins: !expander:PA4

#[neopixel expanderPixel]
#pin: expander:PB1
#chain_count: 1
#initial_RED: 0.9
#initial_GREEN: 0.3
#initial_BLUE: 0.0

# Other Pins
#T0 = PA6
#T1 = PA5
#GPIO = PA7

timmit99, 2024-05-26:
`sensor_pin: PA6`
Needs to be 
`sensor_pin: expander:PA6`

pepegragera, 2024-05-26:
The error is the same
[image]

**Klippy log**
Config error
Traceback (most recent call last):
  File "/home/pepgra/klipper/klippy/klippy.py", line 175, in _connect
    self._read_config()
  File "/home/pepgra/klipper/klippy/klippy.py", line 143, in _read_config
    m.add_printer_objects(config)
  File "/home/pepgra/klipper/klippy/toolhead.py", line 670, in add_printer_objects
    kinematics.extruder.add_printer_objects(config)
  File "/home/pepgra/klipper/klippy/kinematics/extruder.py", line 310, in add_printer_objects
    pe = PrinterExtruder(config.getsection(section), i)
  File "/home/pepgra/klipper/klippy/kinematics/extruder.py", line 140, in __init__
    self.heater = pheaters.setup_heater(config, gcode_id)
  File "/home/pepgra/klipper/klippy/extras/heaters.py", line 275, in setup_heater
    self.register_sensor(config, heater, gcode_id)
  File "/home/pepgra/klipper/klippy/extras/heaters.py", line 300, in register_sensor
    raise self.printer.config_error(
configparser.Error: G-Code sensor id T0 already registered
webhooks client 547985549344: New connection
webhooks client 547985549344: Client info {'program': 'Moonraker', 'version': 'v0.8.0-373-gc958120'}

timmit99, 2024-05-26:
That error is different. That error says that the name T0 is already registered (in use). 
Change the Gcode id to something new or comment out that line.

pepegragera, 2024-05-27:
Thanks! all ok

[temperature_sensor Interior]
sensor_type: ATC Semitec 104GT-2
sensor_pin: expander:PA6
#gcode_id: T0