VoronDesign/Voron-Hardware #231: Klipper Expander - Update Optional Features Doc

louisrousseau, 2025-02-06:
The `Support external sensor devices` option was split in https://github.com/Klipper3d/klipper/commit/1499bfa4892908bc63c61969aba0b9d64f078f74 and "Optional features" should (as far as I understand) now look like this: 

[image]

yebo29, 2025-02-27:
Can confirm this worked for me as well trying to make for the STM32 V0 display USB adapter board from the SIBOOR Switchwire kit. I was able to `make` with these exact options.

53Aries, 2025-02-27:
Confirmed working for me also

kitchung, 2025-04-18:
The "Optional features" menu changed again!

[image]

From Klipper v0.13.0-17-g017371b7

thijstriemstra, 2025-04-18:
> I have no idea what they should be; this screenshot is the default settings when I entered this menu.

Yea, since upgrading to klipper v0.13.0-33-g413ff19ea I'm unable to get the klipper expander to work, after flashing it doesn't show up anymore using `lsusb`..

thijstriemstra, 2025-12-07:
i have no idea why nobody bothered to review or merge this PR. pretty disheartening voron